libimobiledevice/usbmuxd #226: Cannot pair iPod 4G anymore.

tester1121, 2023-08-26:
I am running EndeavourOS on my desktop, and I cannot connect to my iPod 4G in any way. All of the idevice tools tell me that there is no device found, and ifuse keeps telling me to unplug and plug back in my iPod. KDE was able to report an error once about lockdown -5.

Console log from `sudo usbmuxd -f -v`:
```
[18:47:45.023][4] Client 17 accepted: usbmuxd[83752]
[18:47:45.023][4] Client 17 connection closed
[18:47:45.023][4] Client 17 is going to be disconnected: usbmuxd[83752]
[18:47:45.139][4] Client 17 accepted: usbmuxd[83752]
[18:47:45.140][4] Client 17 connection closed
[18:47:45.140][4] Client 17 is going to be disconnected: usbmuxd[83752]
[18:47:45.140][4] Client 17 accepted: usbmuxd[83752]
[18:47:45.140][4] Client 17 connection closed
[18:47:45.140][4] Client 17 is going to be disconnected: usbmuxd[83752]
[18:47:45.142][4] Client 17 accepted: usbmuxd[83752]
[18:47:45.142][4] Client 17 connection closed
[18:47:45.142][4] Client 17 is going to be disconnected: usbmuxd[83752]
[18:47:45.244][1] preflight_worker_handle_device_add: ERROR StartSession failed on device (UDID of what I think is my iPod), lockdown error -5
[18:47:45.250][4] Connection reset by device 1 (2->62078)
[18:47:45.250][4] Client 15 is going to be disconnected: usbmuxd[83752]
[18:47:45.250][4] device_control_input: AppleUSBDeviceMux::sessionUpcall socket is closed, session 0x987553c8 (62078<-lo0->49261 62078<-usb->2)
[18:47:49.840][4] Client 14 accepted: /usr/lib/upowerd[5951]
[18:47:49.840][4] Client 14 connection closed
[18:47:49.840][4] Client 14 is going to be disconnected: /usr/lib/upowerd[5951]
[18:47:54.846][4] Client 14 accepted: /usr/lib/upowerd[5951]
[18:47:54.846][4] Client 14 connection closed
[18:47:54.846][4] Client 14 is going to be disconnected: /usr/lib/upowerd[5951]
``` 

The UDID is also in my idevicepair list of paired UDIDs, but I have never successfully paired the iPod through idevicepair. I also cannot unpair the UDID.

closed 2023-09-11 by tester1121

ThatDumbPan, 2025-07-29:
Did you ever find a solution

hackysphere, 2025-07-29:
No, I think I just ended up using a separate Windows computer.