libimobiledevice/usbmuxd #230: udev rules do not match devices in recovery/DFU mode

apsun, 2023-09-29:
While using `idevicerestore` I noticed that it would fail with `LIBUSB_ERROR_ACCESS` unless run as root. It looks like when the device is rebooted into recovery mode, the udev rules no longer match and so the files are not owned by the `usbmux` user, causing the command to fail.

From `lsusb`:

Recovery mode:
```
Bus 001 Device 024: ID 05ac:1281 Apple, Inc. Apple Mobile Device [Recovery Mode]
```

DFU mode:
```
Bus 001 Device 025: ID 05ac:1227 Apple, Inc. Mobile Device (DFU Mode)
```

I don't have any other devices to test with, but based on a quick search I found https://devicehunt.com/view/type/usb/vendor/05AC which leads me to believe that the device IDs are the same for all devices in recovery/DFU mode. Would it make sense to add these to https://github.com/libimobiledevice/usbmuxd/blob/master/udev/39-usbmuxd.rules.in?

nikias, 2023-09-29:
This is handled already here: https://github.com/libimobiledevice/libirecovery/blob/master/udev/39-libirecovery.rules.in

apsun, 2023-09-29:
Oh I see, sorry I missed that. It seems like the libirecovery udev rule doesn't set `OWNER="usbmux"`, but instead `OWNER="root", GROUP="storage"` - is that expected?

nikias, 2023-09-29:
Yes. The device file is not supposed to be used by usbmuxd at all, so it doesn't need usbmux permissions. Instead, the current user (or the user that is supposed to use irecovery/idevicerestore) should be in the storage group. That group is default on ubuntu and debian to my knowledge.

apsun, 2023-09-29:
Ah, that explains it. I'm running Arch and never added my user to that group. Although it doesn't look like Ubuntu or Debian automatically add that group either nowadays and instead use systemd + [uaccess](https://wiki.archlinux.org/title/users_and_groups#Pre-systemd_groups) to automatically grant the current user access (that said, I don't have an install on hand to test, and this is all very new to me as well)