libimobiledevice/usbmuxd #233: Can't connect to old devices (iOS 12), device reboot after unplugging

hiepnh0, 2023-10-31:
- Plug in device 
syslog:
```
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.771][3] usbmuxd v1.1.0-158-gf1f8d58 starting up
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.772][3] Successfully dropped privileges to 'usbmux'
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.772][3] Using libusb 1.0.20
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.774][3] Initialization complete
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.774][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.775][2] Switching device 1-26 mode to 3
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.778][2] Cannot find device entry while removing USB device 0x6dfeb0 on location 0x1001a
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.778][2] Cannot find device entry while removing USB device 0x6dfeb0 on location 0x1001a
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.780][3] usbmuxd shutting down
Oct 31 09:27:35 hiep-Vostro-3668 usbmuxd[15371]: [09:27:35.881][3] Shutdown complete
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.399][3] usbmuxd v1.1.0-158-gf1f8d58 starting up
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.405][3] Successfully dropped privileges to 'usbmux'
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.414][3] Using libusb 1.0.20
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][3] Initialization complete
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][3] Found CDC-NCM and Apple USB Multiplexor in device 1-27 configuration 5
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][2] Skipping switch device 1-27 mode from 3 to 3
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: libusb: error [op_get_configuration] device unconfigured
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][3] Found usbmux interface for device 1-27: 1
Oct 31 09:27:36 hiep-Vostro-3668 usbmuxd[15407]: [09:27:36.426][3] Changing configuration of device 1-27: 0 -> 5
Oct 31 09:28:38 hiep-Vostro-3668 usbmuxd[15407]: [09:28:38.885][1] Failed to request lang ID for device 1-27 (2)
```

- Unplug device
syslog:
```
Oct 31 09:29:27 hiep-Vostro-3668 usbmuxd[15407]: [09:29:27.968][2] Cannot find device entry while removing USB device 0xeeb0f0 on location 0x1001b
Oct 31 09:29:27 hiep-Vostro-3668 usbmuxd[15407]: [09:29:27.968][2] Cannot find device entry while removing USB device 0xeeb0f0 on location 0x1001b
Oct 31 09:29:27 hiep-Vostro-3668 usbmuxd[15407]: [09:29:27.985][3] usbmuxd shutting down
Oct 31 09:29:28 hiep-Vostro-3668 usbmuxd[15407]: [09:29:28.086][3] Shutdown complete
```

$usbmuxd --version
usbmuxd 1.1.0-158-gf1f8d58

iPhone 6
iOS 12.5.7

IsanHu, 2023-12-22:
Hello, is there any progress on this issue?

httpstorm, 2024-04-24:
Mode 3 breaks tethering for me. Reverting to mode 1 makes it work again. Here is my research https://github.com/libimobiledevice/usbmuxd/commit/c7a0dd9b82633ea347497626282e3051a469ef50#r141306029

@nikias @FunkyM @BalkanMadman @rickmark @Blefish @marcan

httpstorm, 2024-04-24:
@nikias 
Thanks for the quick replay. Are you able to enable tethering in mode 3 and how?
- The benefit of mode 1 is that we only need `ipheth` so the interface will always be named `eth0` when we switch between old and new iPhones. With mode 3 we would need configurations for both `eth0` and `usb0`.
- The con of mode 1 is slower performance. This could be addressed by optimising `ipheth`.

I will make a PR to update all `libimobiledevice` in OpenWRT shortly. The latest release for some of the packages is 2020-06-16 and it does not compile unless we move forward, so I switched them to the latest commit.

> I was looking to add a way to dynamically allow mode switching on demand, but haven't completed my work on that yet.

If you come up with interesting ideas, I can help you test them.
Can you please have a look at https://github.com/libimobiledevice/usbmuxd/issues/218 ? We addressed it here https://github.com/openwrt/packages/pull/23976
It would be nice if you can fix it in `usbmuxd`.

httpstorm, 2025-08-15:
@Leoaaron 
> I have the same issue, any update?

I use OpenWRT main branch, where all `libimobiledevice` packages are updated either to the latest release, or if the release is a few months old, we use a current snapshot. I have tested the following devices all of which work: iPhone 3G (iOS 4.2.1), 4S (iOS 8.4), 7 Plus (iOS 15.8.4), 11 (iOS 18.6).

Leoaaron, 2025-08-15:
> [@Leoaaron](https://github.com/Leoaaron)
>
>
> I use OpenWRT main branch, where all `libimobiledevice` packages are updated either to the latest release, or if the release is a few months old, we use a [...]

Thanks for your reply, but the system we are using is Ubuntu. Which kernel function is the key to solving this problem? Is it CONFIG_USB_IPHETH? I hope this problem can be solved in the distribution version.