libimobiledevice/usbmuxd #237: Commit c7a0dd9 : usb tethering doesn't work anymore

maravtdm, 2024-01-31:
Hi,

With this commit :
https://github.com/libimobiledevice/usbmuxd/commit/c7a0dd9b82633ea347497626282e3051a469ef50
usb tethering doesn't work anymore
/var/log/messages
```
Feb  1 00:10:53 blackstar kernel: usb 3-2: new high-speed USB device number 40 using xhci_hcd
Feb  1 00:10:53 blackstar kernel: usb 3-2: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=13.03
Feb  1 00:10:53 blackstar kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  1 00:10:53 blackstar kernel: usb 3-2: Product: iPhone
Feb  1 00:10:53 blackstar kernel: usb 3-2: Manufacturer: Apple Inc.
Feb  1 00:10:53 blackstar kernel: usb 3-2: SerialNumber: 000081010003391C1AD0001E
Feb  1 00:10:54 blackstar mtp-probe: checking bus 3, device 40: "/sys/devices/pci0000:00/0000:00:08.1/0000:03:00.4/usb3/3-2" 
Feb  1 00:10:54 blackstar mtp-probe: bus: 3, device: 40 was not an MTP device 
Feb  1 00:10:54 blackstar kernel: apple-mfi-fastcharge 3-2: USB disconnect, device number 40
Feb  1 00:10:54 blackstar kernel: usb 3-2: new high-speed USB device number 41 using xhci_hcd
Feb  1 00:10:54 blackstar kernel: usb 3-2: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=13.03
Feb  1 00:10:54 blackstar kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  1 00:10:54 blackstar kernel: usb 3-2: Product: iPhone
Feb  1 00:10:54 blackstar kernel: usb 3-2: Manufacturer: Apple Inc.
Feb  1 00:10:54 blackstar kernel: usb 3-2: SerialNumber: 000081010003391C1AD0001E
Feb  1 00:10:54 blackstar mtp-probe: checking bus 3, device 41: "/sys/devices/pci0000:00/0000:00:08.1/0000:03:00.4/usb3/3-2" 
Feb  1 00:10:54 blackstar mtp-probe: bus: 3, device: 41 was not an MTP device
```
/var/log/syslog
```
Feb  1 00:10:54 blackstar usbmuxd[14461]: [2] Switching device 3-40 mode to 3 
Feb  1 00:10:54 blackstar usbmuxd[14461]: [2] Cannot find device entry while removing USB device 0x10ffbd0 on location 0x30028 
```

Before, usb tethering works fine
commit:
https://github.com/libimobiledevice/usbmuxd/commit/91aa7be081d835244f2118f582e8e77988d74c58
/var/log/messages when plugged in
```
Feb  1 00:15:59 blackstar kernel: usb 3-2: new high-speed USB device number 42 using xhci_hcd
Feb  1 00:15:59 blackstar kernel: usb 3-2: New USB device found, idVendor=05ac, idProduct=12a8, bcdDevice=13.03
Feb  1 00:15:59 blackstar kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  1 00:15:59 blackstar kernel: usb 3-2: Product: iPhone
Feb  1 00:15:59 blackstar kernel: usb 3-2: Manufacturer: Apple Inc.
Feb  1 00:15:59 blackstar kernel: usb 3-2: SerialNumber: 000081010003391C1AD0001E
Feb  1 00:15:59 blackstar mtp-probe: checking bus 3, device 42: "/sys/devices/pci0000:00/0000:00:08.1/0000:03:00.4/usb3/3-2" 
Feb  1 00:15:59 blackstar mtp-probe: bus: 3, device: 42 was not an MTP device 
Feb  1 00:16:00 blackstar kernel: ipheth 3-2:4.2: ipheth_enable_ncm: usb_control_msg: 0
Feb  1 00:16:00 blackstar kernel: ipheth 3-2:4.2: Apple iPhone USB Ethernet device attached
Feb  1 00:16:00 blackstar NetworkManager[1084]: <info>  [1706742960.0476] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/21)
Feb  1 00:16:00 blackstar NetworkManager[1084]: <info>  [1706742960.0526] device (eth0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3288] device (eth0): carrier: link connected
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3289] device (eth0): state change: unavailable -> disconnected (reason 'carrier-changed', sys-iface-state: 'managed')
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3294] policy: auto-activating connection 'eth0' (fe9feb68-c34d-3748-aa79-f97c85697a85)
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3297] device (eth0): Activation: starting connection 'eth0' (fe9feb68-c34d-3748-aa79-f97c85697a85)
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3297] device (eth0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3299] device (eth0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3302] device (eth0): state change: config -> ip-config (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3306] dhcp4 (eth0): activation: beginning transaction (timeout in 45 seconds)
Feb  1 00:16:01 blackstar NetworkManager[1084]: <info>  [1706742961.3327] dhcp4 (eth0): dhcpcd started with pid 19291
Feb  1 00:16:01 blackstar dhcpcd[19291]: dhcpcd-10.0.6 starting
Feb  1 00:16:01 blackstar dhcpcd[19292]: DUID 00:01:00:01:2d:3c:fb:17:a2:fb:c5:dc:b7:b5
Feb  1 00:16:01 blackstar dhcpcd[19292]: eth0: IAID c5:dc:b7:4a
Feb  1 00:16:02 blackstar ModemManager[1218]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:08.1/0000:03:00.4/usb3/3-2': not supported by any plugin 
Feb  1 00:16:02 blackstar dhcpcd[19292]: eth0: rebinding lease of 172.20.10.3
Feb  1 00:16:02 blackstar dhcpcd[19292]: eth0: leased 172.20.10.3 for 86400 seconds
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7260] dhcp4 (eth0):   address 172.20.10.3
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7260] dhcp4 (eth0):   plen 28 (255.255.255.240)
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7260] dhcp4 (eth0):   gateway 172.20.10.1
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7260] dhcp4 (eth0):   lease time 86400
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7260] dhcp4 (eth0):   nameserver '172.20.10.1'
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7261] dhcp4 (eth0): state changed new lease, address=172.20.10.3
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7267] policy: set 'eth0' (eth0) as default for IPv4 routing and DNS
Feb  1 00:16:02 blackstar dbus-daemon[1049]: [system] Activating service name='org.freedesktop.nm_dispatcher' requested by ':1.1' (uid=0 pid=1084 comm="/usr/sbin/NetworkManager") (using servicehelper)
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7318] device (eth0): state change: ip-config -> ip-check (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:02 blackstar dbus-daemon[1049]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7357] device (eth0): state change: ip-check -> secondaries (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7359] device (eth0): state change: secondaries -> activated (reason 'none', sys-iface-state: 'managed')
Feb  1 00:16:02 blackstar NetworkManager[1084]: <info>  [1706742962.7362] device (eth0): Activation: successful, device activated.
Feb  1 00:16:03 blackstar NetworkManager[1084]: <info>  [1706742963.4393] dhcp6 (eth0): activation: beginning transaction (timeout in 45 seconds)
Feb  1 00:16:03 blackstar NetworkManager[1084]: <info>  [1706742963.4398] policy: set 'eth0' (eth0) as default for IPv6 routing and DNS
Feb  1 00:16:03 blackstar NetworkManager[1084]: <info>  [1706742963.4464] dhcp6 (eth0): state changed new lease
```

/var/log/syslog
```
Feb  1 00:15:59 blackstar usbmuxd[19266]: [2] Skipping switch device 3-42 mode from 1 to 1 

```

OS : Slackware
Kernel : 6.7.2
iPhone : 12 Pro
iOS : 17.3

Mark6222, 2024-02-01:
ive the same issue

Mark6222, 2024-02-01:
i think its something to do with the udev rules
`make[2]: Entering directory '/home/mark/usbipserver/usbmuxd/udev'
/bin/sed -r -e 's|@udev_activation_rule[@]|ENV{SYSTEMD_WANTS}="usbmuxd.service"|g' -e 's|@sbindir[@]|/usr/sbin|g' < 39-usbmuxd.rules.in > 39-usbmuxd.rules || rm 39-usbmuxd.rules
make[2]: Leaving directory '/home/mark/usbipserver/usbmuxd/udev'
Making all in systemd
make[2]: Entering directory '/home/mark/usbipserver/usbmuxd/systemd'
/bin/sed -r -e 's|@sbindir[@]|/usr/sbin|g' -e 's|@runstatedir[@]|/run|g' < usbmuxd.service.in > usbmuxd.service || rm usbmuxd.service
make[2]: Leaving directory '/home/mark/usbipserver/usbmuxd/systemd'
Making all in docs`

kumama, 2024-02-02:
Adding below to systemd unit help this
Environment=USBMUXD_DEFAULT_DEVICE_MODE=1

Mark6222, 2024-02-07:
> Adding below to systemd unit help this Environment=USBMUXD_DEFAULT_DEVICE_MODE=1

sorry i dont understand where do i put this

maravtdm, 2024-02-07:
> > Adding below to systemd unit help this Environment=USBMUXD_DEFAULT_DEVICE_MODE=1
> 
> sorry i dont understand where do i put this

Anyway, it doesn't help that much on non-systemd distributions

JonasVautherin, 2024-08-17:
Isn't this the same thing as https://github.com/libimobiledevice/usbmuxd/issues/252, which has an [open PR](https://github.com/libimobiledevice/usbmuxd/pull/242)? At least #242 solves my issue which looks very similar.

pelaajahacks, 2024-09-01:
> Adding below to systemd unit help this Environment=USBMUXD_DEFAULT_DEVICE_MODE=1

It works! Thank you so much, ive been trying to fix this problem for weeks.🥹

httpstorm, 2024-09-16:
The default mode is changed back to 1, so it should work now https://github.com/libimobiledevice/usbmuxd/commit/f8be60c4a3c34ddcd97a6b0481ef0e81214cecc5