libimobiledevice/usbmuxd #247: usbmuxd logging asyncReadComplete, message was too large (65536 bytes, max = 65535) while using usbipd-win

d0n13, 2024-05-24:
Hi all

I'm trying to run `idevicerestore --erase --latest` on WSL using the [usbipd-win](https://github.com/dorssel/usbipd-win) project but I'm getting the following error logged in syslog on WSL.

```
2024-05-23T16:41:42.806058+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.805][3] usbmuxd v1.1.1 starting up
2024-05-23T16:41:42.806434+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.806][3] Successfully dropped privileges to 'usbmux'
2024-05-23T16:41:42.806483+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.806][3] Using libusb 1.0.27
2024-05-23T16:41:42.808529+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: libusb: warning [op_get_configuration] device unconfigured
2024-05-23T16:41:42.808604+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: libusb: error [op_get_active_config_descriptor] device unconfigured
2024-05-23T16:41:42.808635+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.808][3] Could not get old configuration descriptor for device 1-21: LIBUSB_ERROR_NOT_FOUND
2024-05-23T16:41:42.850999+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.850][3] Initialization complete
2024-05-23T16:41:42.851142+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.850][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
2024-05-23T16:41:42.852935+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.852][3] Connecting to new device on location 0x10015 as ID 1
2024-05-23T16:41:42.853395+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:42.853][3] Connected to v2.0 device 1 on location 0x10015 with serial number 00008101-001E48CXXXXXXXXX
2024-05-23T16:41:48.692940+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Daemon: connecting to Windows Agent
2024-05-23T16:41:48.693071+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Connecting"
2024-05-23T16:41:48.693316+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[33mWARNING#033[0m Daemon: could not connect to Windows Agent: could not get address: could not read agent port file "/mnt/c/Users/donie/.ubuntupro/.address": open /mnt/c/Users/donie/.ubuntupro/.address: no such file or directory
2024-05-23T16:41:48.693382+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Reconnecting to Windows host in 60 seconds
2024-05-23T16:41:48.693406+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Not connected: waiting to retry"
2024-05-23T16:41:49.409533+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:49.409][1] device_control_input: ERROR (on device): asyncReadComplete, message was too large (65536 bytes, max = 65535)
2024-05-23T16:41:49.411258+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:41:49.411][1] device_control_input: ERROR (on device): asyncReadComplete, message was too large (65536 bytes, max = 65535)
2024-05-23T16:42:23.510946+01:00 DESKTOP-LVPVFTP usbmuxd[58211]: [16:42:23.510][2] device_control_input: Got unhandled payload type 4
2024-05-23T16:42:48.752655+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Daemon: connecting to Windows Agent
2024-05-23T16:42:48.757790+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Connecting"
2024-05-23T16:42:48.757865+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[33mWARNING#033[0m Daemon: could not connect to Windows Agent: could not get address: could not read agent port file "/mnt/c/Users/donie/.ubuntupro/.address": open /mnt/c/Users/donie/.ubuntupro/.address: no such file or directory
2024-05-23T16:42:48.757878+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[36mINFO#033[0m Reconnecting to Windows host in 60 seconds
2024-05-23T16:42:48.757888+01:00 DESKTOP-LVPVFTP wsl-pro-service[208]: #033[37mDEBUG#033[0m Updated systemd status to "Not connected: waiting to retry"
```

In my setup I have an iPhone bound and attached to WSL from Window and the idevicerestore is failing once it starts to transfer data to the device.

I have also noticed this same error if we try to sideload and app to an iOS device.

Seems there is an issue with message sizes while transferring via usbip. Any possibility somebody could look into this who understands the flow?

The issue was raised with the [usbipd-win](https://github.com/dorssel/usbipd-win) project [here ](https://github.com/dorssel/usbipd-win/issues/959)but they guys there seem to think this may be a usbmuxd problem. Thanks to anyone who has time to look at this.

Donie

nikias, 2024-05-24:
Why don't you install Apple's AppleMobileDeviceSupport package (part of iTunes installer)? It comes with the official usbmuxd and then you don't have to compile usbmuxd.

d0n13, 2024-05-24:
> Why don't you install Apple's AppleMobileDeviceSupport package (part of iTunes installer)? It comes with the official usbmuxd and then you don't have to compile the open source usbmuxd.

But usbmuxd runs on the WSL side under Linux? How does it work if the usbmuxd is under Windows?

nikias, 2024-05-24:
I have no idea, it has an open socket 27015, you can try if you can connect. Apart from that you can build libimobiledevice and tools for Windows using msys2.

d0n13, 2024-05-24:
> I have no idea, it has an open socket 27015, you can try if you can connect. Apart from that you can build libimobiledevice and tools for Windows using msys2.

I did try to build using mysys2 but it failed in linking stage even though I had build everything correctly or so I thought. Is there correct build instruction for mysys2 anywhere?

I got this error when building libimobiledevice and I couldn't see what I was missing...

```
  CCLD     libimobiledevice-1.0.la
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ../3rd_party/libsrp6a
-sha512/.libs/libsrp6a-sha512.a(t_misc.o):t_misc.c:(.rdata$.refptr.__imp__environ[.refptr.__imp__environ]+0x0): undefined r
eference to `__imp__environ'
collect2.exe: error: ld returned 1 exit status
```