libimobiledevice/usbmuxd #252: iPhone tethering does not work after the default mode change from 1 to 3

httpstorm, 2024-07-19:
[1] changes to mode 3 CDC NCM by default. In this mode the iPhone exposes an Ethernet interface, handled by `kmod-usb-net-cdc-ncm`. This mode is intended for reverse tethering, where the computer shares its Internet connection with an iPhone. The iPhone is a DHCP client.

Since most people expect to use tethering, where the iPhone shares its Internet connection with a computer, it would be more appropriate to revert default mode back to 1 [2]. With mode 1, the `kmod-usb-net-ipheth` driver is used, where support for CDC NCM is available since Linux kernel 6.6, and iPhones are automatically configured in this mode.

Description of different modes
https://www.synacktiv.com/en/publications/ios-a-journey-in-the-usb-networking-stack

[1] https://github.com/libimobiledevice/usbmuxd/commit/c7a0dd9b82633ea347497626282e3051a469ef50
[2] Revert default mode back to 1 (PR): https://github.com/libimobiledevice/usbmuxd/pull/242

Any comments?
@nikias @marcan @qmfrederik @FunkyM @BalkanMadman @Blefish 
@Forst @mbilker

Forst, 2024-07-19:
Mode 3 indeed does not support internet sharing from phone to computer, as in this mode "AppleUSBEthernet" interfaces are not exposed in any available configuration. Mode 3 only has reverse tethering available. This article [1] has a good overview.

For compatibility with older iOS versions, I recommend sticking to mode 1.

Newer iOS versions have mode 4, which supports both regular and reverse tethering simultaneously. However this mode is not yet supported by the `ipheth` kernel module in Linux.

The past couple of weeks I've been working on updating the `ipheth` module to support mode 4, `usbmuxd` minimal support for modes 4 and 5, and documenting the different available modes, configurations and interfaces. I can post WIPs if there is interest, but I'd rather clean them up a bit first 🙂 

[1]: https://www.synacktiv.com/en/publications/ios-a-journey-in-the-usb-networking-stack

httpstorm, 2024-09-16:
Fixed by this commit:
usb: Revert default mode back to 1 https://github.com/libimobiledevice/usbmuxd/commit/f8be60c4a3c34ddcd97a6b0481ef0e81214cecc5

closed 2024-09-16 by httpstorm