libimobiledevice/usbmuxd #253: usbmuxd not installed properly

MeevTheGigachad, 2024-08-01:
The version of usbmuxd I'm trying to install is version 1.1.1-58-gbc0b91c, taken directly from source.
The installation probably failed at `sudo make install` step where I did not see a string mentioning the library was installed into /usr/local/

For example, if I were to build and install libtatus, `sudo make install` would give me:

`Making install in src`
`make[1]: Entering directory '/home/meev/libtatsu/src'`
`make[2]: Entering directory '/home/meev/libtatsu/src'`
 `/usr/bin/mkdir -p '/usr/local/lib'`
` /bin/bash ../libtool   --mode=install /usr/bin/install -c   libtatsu.la '/usr/local/lib'`
`libtool: install: /usr/bin/install -c .libs/libtatsu.so.0.0.1 /usr/local/lib/libtatsu.so.0.0.1`
`libtool: install: (cd /usr/local/lib && { ln -s -f libtatsu.so.0.0.1 libtatsu.so.0 || { rm -f libtatsu.so.0 && ln -s libtatsu.so.0.0.1 libtatsu.so.0; }; })`
`libtool: install: (cd /usr/local/lib && { ln -s -f libtatsu.so.0.0.1 libtatsu.so || { rm -f libtatsu.so && ln -s libtatsu.so.0.0.1 libtatsu.so; }; })`
`libtool: install: /usr/bin/install -c .libs/libtatsu.lai /usr/local/lib/libtatsu.la`
`libtool: install: /usr/bin/install -c .libs/libtatsu.a /usr/local/lib/libtatsu.a`
`libtool: install: chmod 644 /usr/local/lib/libtatsu.a`
`libtool: install: ranlib /usr/local/lib/libtatsu.a`
`libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib`
`----------------------------------------------------------------------`
`Libraries have been installed in:`
  ` /usr/local/lib`

`If you ever happen to want to link against installed libraries`
`in a given directory, LIBDIR, you must either use libtool, and`
`specify the full pathname of the library, or use the '-LLIBDIR'`
`flag during linking and do at least one of the following:`
  ` - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution`
`   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking`
`   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag`
`   - have your system administrator add LIBDIR to '/etc/ld.so.conf'`

`See any operating system documentation about shared libraries for`
`more information, such as the ld(1) and ld.so(8) manual pages.`
`---------------------------------------------------------------------`
` /usr/bin/mkdir -p '/usr/local/lib/pkgconfig'`
` /usr/bin/install -c -m 644 libtatsu-1.0.pc '/usr/local/lib/pkgconfig'`
`make[2]: Leaving directory '/home/meev/libtatsu/src'`
`make[1]: Leaving directory '/home/meev/libtatsu/src'`
`Making install in include`
`make[1]: Entering directory '/home/meev/libtatsu/include'`
`make[2]: Entering directory '/home/meev/libtatsu/include'`
`make[2]: Nothing to be done for 'install-exec-am'.`
 `/usr/bin/mkdir -p '/usr/local/include'`
` /usr/bin/mkdir -p '/usr/local/include/libtatsu'`
` /usr/bin/install -c -m 644  libtatsu/tss.h libtatsu/tatsu.h '/usr/local/include/libtatsu'`
`make[2]: Leaving directory '/home/meev/libtatsu/include'`
`make[1]: Leaving directory '/home/meev/libtatsu/include'`
`make[1]: Entering directory '/home/meev/libtatsu'`
`make[2]: Entering directory '/home/meev/libtatsu'`
`make[2]: Nothing to be done for 'install-exec-am'.`
`make[2]: Nothing to be done for 'install-data-am'.`
`make[2]: Leaving directory '/home/meev/libtatsu'`
`make[1]: Leaving directory '/home/meev/libtatsu'`

But weirdly, when I `sudo make install` usbmuxd, it gave me:

`Making install in src`
`make[1]: Entering directory '/home/meev/usbmuxd/src'`
`make[2]: Entering directory '/home/meev/usbmuxd/src'`
` /usr/bin/mkdir -p '/usr/local/sbin'`
`  /bin/bash ../libtool   --mode=install /usr/bin/install -c usbmuxd '/usr/local/sbin'`
`libtool: install: /usr/bin/install -c usbmuxd /usr/local/sbin/usbmuxd`
`make[2]: Nothing to be done for 'install-data-am'.`
`make[2]: Leaving directory '/home/meev/usbmuxd/src'`
`make[1]: Leaving directory '/home/meev/usbmuxd/src'`
`Making install in udev`
`make[1]: Entering directory '/home/meev/usbmuxd/udev'`
`make[2]: Entering directory '/home/meev/usbmuxd/udev'`
`make[2]: Nothing to be done for 'install-exec-am'.`
 `/usr/bin/mkdir -p '/lib/udev/rules.d'`
 `/usr/bin/install -c -m 644 39-usbmuxd.rules '/lib/udev/rules.d'`
`make[2]: Leaving directory '/home/meev/usbmuxd/udev'`
`make[1]: Leaving directory '/home/meev/usbmuxd/udev'`
`Making install in systemd`
`make[1]: Entering directory '/home/meev/usbmuxd/systemd'`
`make[2]: Entering directory '/home/meev/usbmuxd/systemd'`
`make[2]: Nothing to be done for 'install-exec-am'.`
 `/usr/bin/mkdir -p '/lib/systemd/system'`
 `/usr/bin/install -c -m 644 usbmuxd.service '/lib/systemd/system'`
`make[2]: Leaving directory '/home/meev/usbmuxd/systemd'`
`make[1]: Leaving directory '/home/meev/usbmuxd/systemd'`
`Making install in docs`
`make[1]: Entering directory '/home/meev/usbmuxd/docs'`
`make[2]: Entering directory '/home/meev/usbmuxd/docs'`
`make[2]: Nothing to be done for 'install-exec-am'.`
 `/usr/bin/mkdir -p '/usr/local/share/man/man8'`
 `/usr/bin/install -c -m 644 usbmuxd.8 '/usr/local/share/man/man8'`
`make[2]: Leaving directory '/home/meev/usbmuxd/docs'`
`make[1]: Leaving directory '/home/meev/usbmuxd/docs'`
`make[1]: Entering directory '/home/meev/usbmuxd'`
`make[2]: Entering directory '/home/meev/usbmuxd'`
`make[2]: Nothing to be done for 'install-exec-am'.`
`make[2]: Nothing to be done for 'install-data-am'.`
`make[2]: Leaving directory '/home/meev/usbmuxd'`
`make[1]: Leaving directory '/home/meev/usbmuxd'`

Notice the line `Libraries have been installed in:` is missing from the above result.
The operating system executing the commands above is Debian 12.6 if it's relevant.

mexmer, 2024-08-01:
usbmuxd doesn't have libraries it's service binary.

mexmer, 2024-08-01:
if you need to interface with it, you need to install libusbmuxd, and libimobiledevice, for more functionality.

MeevTheGigachad, 2024-08-01:
Sorry, the information above is not valid anymore. The problem was I waited out for too long so my MacBook automatically returns to recovery mode and errors out. After that I just redo the procedure a few more times until it worked and held a stable connection to my MacBook.

closed 2024-08-01 by MeevTheGigachad