libimobiledevice/usbmuxd #261: Clarify License in use

klaus-halfmann, 2025-02-06:
Somewhere between Version [1.1.0](https://github.com/libimobiledevice/usbmuxd/releases/tag/1.1.0)
and  [1.1.1](https://github.com/libimobiledevice/usbmuxd/releases/tag/1.1.1)

The plain README was exchange with a [Readme.md](https://github.com/libimobiledevice/usbmuxd/blob/master/README.md)

The new README states: 
> This library and utilities are licensed under the [GNU General Public License v3.0](https://www.gnu.org/licenses/gpl-3.0.en.html), also included in the repository in the COPYING.GPLv3 file.

However the GPL v3.0 and v.2.0 are both still in the Repo and are both visible in the github UI.

As there are some major difference between v2.0 and v3.0 anyone using this tool in a "closed" device
will have a risc that some owner of such a device will enforce the opening of the device to the public.

So either:
* Remove any references to the GPL like [COPYING.GPLv2](https://github.com/libimobiledevice/usbmuxd/blob/master/COPYING.GPLv2)
OR
* state that usbmux can be used under the Terms or the GPLv2 in the [Readme.md](https://github.com/libimobiledevice/usbmuxd/blob/master/README.md)

On https://libimobiledevice.org/#get-started It is stated [usbmuxd](https://libimobiledevice.org/#download-details-usbmuxd) [1.1.1](https://github.com/libimobiledevice/usbmuxd/releases/tag/1.1.1) uses http://choosealicense.com/licenses/gpl-2.0. which makes things even more confusing

I found no other sources related to this change in License, please point me to any discussions that may apply.

MacOS, 2025-02-14:
Hi all,

first, thank you to the team behind this project for their work!

I'm also interested in this issue. May I add to the issue opener, that the situation is even more complex. It is for example the case that the comments state that at least parts of the source code is dual-licenses, meaning one can choose the license between GPL-v2 and GPL-v3, see here

https://github.com/libimobiledevice/usbmuxd/blob/523f7004dce885fe38b4f80e34a8f76dc8ea98b5/src/client.c#L1-L19

I would suggest to add ``SPDX`` License Identifiers to each source code file. The linked source code header would then become

```c
/*
 * client.c
 *
 * Copyright (C) 2009 Hector Martin <hector@marcansoft.com>
 * Copyright (C) 2009 Nikias Bassen <nikias@gmx.li>
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 2 or version 3.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA
 *
 * SPDX-License-Identifier: GPL-2.0-or-later
 * SPDX-FileCopyrightText: 2009 Hector Martin <hector@marcansoft.com>, 2009 Nikias Bassen <nikias@gmx.li>
 * SPDX-FileContributor: Hector Martin <hector@marcansoft.com>, Nikias Bassen <nikias@gmx.li>
 * SPDX-FileType: SOURCE
 */
```

I'm happy to take this issue over and insert the license header information in all source files. In addition, I am also offering to make the repository [``REUSE``](https://reuse.software/) compliant. If so, please assign this issue to me.

Thank you!

MacOS, 2025-03-07:
Hi @nikias,

thank you for your work on this library! 

May I ask you to answer on this issue as soon as you can afford it because we have a high interest in addressing this issue.

Thank you in advance!