libimobiledevice/usbmuxd #263: Compile error in Windows

alvin1995, 2025-05-15:
When I used MSYS2 to compile usbmuxd on Windows(Before this I used ./autogen.sh and succeeded)
I encountered the following header file error.

[image]

Maybe winsock2.h should be used on Windows.
There are many similar compilation problems. I think someone may have encountered the same problem as me.
How to solve it? Looking forward to your reply, thank you

NumberOneGit, 2025-09-20:
@MarijnS95 Is that the case for ARM64 Windows?   In other words, does Windows still have usbmuxd even though it doesn’t have Apple Mobile Device Ethernet Adapter?

MarijnS95, 2025-10-02:
@NumberOneGit I hadn't considered whether iTunes is distributed for ARM64 Windows. Regardless, we've been considering to fix/port this project to compile and run on Windows to replace the need for a full iTunes install, and that could solve the missing iTunes for ARM64 Windows as well.

nikias, 2025-10-02:
I have a patch for Windows that would allow to compile it, but when run I get an error from libusb saying device hotplug feature is not available, so it would need to use polling. However Windows is then not able to perform it with polling instead.

https://gist.github.com/nikias/c544814c5535d08ab66f0a42afc525ae

Maybe you can get further with this.

nikias, 2025-10-20:
Ok cool, that is at least something. For usb.c I'd suggest to just add an ifdef:
```diff
diff --git a/src/usb.c b/src/usb.c
index d0e9afc..1e3b1f4 100644
--- a/src/usb.c
+++ b/src/usb.c
@@ -527,6 +525,7 @@ static int set_valid_configuration(struct libusb_device* dev, struct usb_device
                }
                if(current_config == 0 || config->bConfigurationValue != current_config) {
                        usbmuxd_log(LL_NOTICE, "Changing configuration of device %i-%i: %i -> %i", bus, address, current_config, config->bConfigurationValue);
+#ifndef _WIN32
                        for(k=0 ; k < config->bNumInterfaces ; k++) {
                                const struct libusb_interface_descriptor *intf1 = &config->interface[k].altsetting[0];
                                if((res = libusb_kernel_driver_active(handle, intf1->bInterfaceNumber)) < 0) {
@@ -541,6 +540,7 @@ static int set_valid_configuration(struct libusb_device* dev, struct usb_device
                                        }
                                }
                        }
+#endif
                        if((res = libusb_set_configuration(handle, j)) != 0) {
                                usbmuxd_log(LL_WARNING, "Could not set configuration %d for device %d-%d: %s", j, bus, address, libusb_error_name(res));
                                libusb_free_config_descriptor(config);
```

This would go directly and call `libusb_set_configuration`. Does that work?

Amoystyle, 2025-10-21:
@nikias 
It doesn't work. It always returns LIBUSB_ERROR_BUSY. This is probably because libusb_kernel_driver_active and libusb_detach_kernel_driver are not supported on Windows. Here's a workaround I used. I hope it helps with Windows support.

[Windows_Support.zip](https://github.com/user-attachments/files/23012179/Windows_Support.zip)

NumberOneGit, 2025-11-03:
@Amoystyle Was this meant to be a workaround to change the USB configuration in Windows?  I see that there was originally an attachment but the comment was edited to remove it.