libimobiledevice/usbmuxd #264: Issue: usbmuxd Consumes 100% CPU When iPhone is Plugged In

dimanech, 2025-05-20:
When I plug my iPhone 12 mini into my Fedora Linux 42 machine via USB, the usbmuxd process starts consuming 100% CPU.

This happens even without pairing the device — it’s only being charged. The issue persists every time the device is plugged in.
When I unplug my device process still running and eating 100% CPU.

## Steps to Reproduce

- Boot into Fedora Linux 42.
- Plug in iPhone 12 mini (iOS 18.4.1) using USB.
- Observe CPU usage: usbmuxd immediately spikes to 100%.

## Expected Behavior

usbmuxd should not consume excessive CPU when an iOS device is plugged in but not paired.

## Actual Behavior

usbmuxd consistently uses 100% CPU while the iPhone is connected via USB, even without any user interaction or pairing.

## System Information

- OS: Fedora Linux 42 (Linux `6.14.6-300.fc42.x86_64`)
- Device: iPhone 12 mini
- iOS Version: `18.4.1`
- usbmuxd version: `v1.1.1^20240915git0b1b233`

## Additional info

When I kill the process and start it in debug mode it do not eating 100% CPU, instead got this error in the loop
```
[23:28:32.891][4] Client 12 accepted: /usr/libexec/upowerd[933]
[23:28:32.891][1] ERROR: Failed to read '/var/lib/lockdown/00008101-00190D662613001E.plist': No such file or directory
[23:28:32.891][4] Client 12 connection closed
[23:28:32.891][4] Client 12 is going to be disconnected: /usr/libexec/upowerd[933]
[23:28:32.982][4] Client 12 accepted: /usr/libexec/upowerd[933]
[23:28:32.982][4] Client 12 connection closed
[23:28:32.982][4] Client 12 is going to be disconnected: /usr/libexec/upowerd[933]
[23:28:32.993][4] Client 13 is going to be disconnected: /usr/libexec/upowerd[933]
```

mexmer, 2025-05-21:
try to replicate it with upowerd disabled, there was case in past, when upowerd was continuously replugging (i mean software way),  usb port used by apple device and did interfere with proper usbmuxd communication.

dimanech, 2025-06-02:
Hi @mexmer thank you for your reply. Actually I do not init this process and even have no idea who do this since it is a part of the GNOME system. Could you please provide more instruction on how to replicate it with upowerd disabled?

mexmer, 2025-06-03:
just stop upowerd process.

Barkbit87, 2025-12-18:
I also have this issue.
I can temporarily fix it by unplug the phone, killing the `usbmuxd` process and then plug in the phone. When plugging in, GNOME shows a message prompting me to unlock my phone.

> Hello, have you reported this issue to https://gitlab.freedesktop.org/upower/upower/-/issues ? I searched `usbmuxd` in it and I found nothing.

This seems to be isolated to `usbmuxd`and not upower...